xelalexv/suniversal #4: Support for Type 5 model S5 + 100

NiklausHofer, 2019-03-30:
I recently got my hands on a SUN Type 5 keyboard with model number "S5 + 100" (see also [this picture of the label plate](https://oc.honet.ch/index.php/s/kSNDRXcckLBmeja#//IMG_20190320_001555.jpg) on the bottom of the keyboard). This keyboard appears to be different from the "Type 5" and "Type 5c" in a number of ways. Most importantly it features Cherry MX Black switches instead of rubber dome. It also does not have the DIP switches exposed to the user. Instead there are jumper cables directly solthered to they keyboard's mainboard. Like on the "Type 5c" the connectors are not directly solthered to the mainboard but connected via a 6 pin connector, very similar to what is shown in your picture. The startup routine of the keyboard may be different as well. All LEDs on the keyboard briefly flash and the keyboard beeps immediately upon receiving current even if only 5V and ground are connected to the keyboard.

I got the same Arduino you used in the development of this project, the Sparkfun Pro Micro 5V. I am certain that I have the 5V variant and not the 3.3V one. It says so on the box and also on the board itself. Also it shows up as `1b4f:9206` in `lsusb` and according to [the SparkFun documentation](https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide/troubleshooting-and-faq) that one is the correct identifier for the 5V variant.

After loading the code onto the Arduino and then re-connecting it, it shows up as USB HID:

`lsusb`:
```
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 0a5c:21e6 Broadcom Corp. BCM20702 Bluetooth 4.0 [ThinkPad]
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 112: ID 1b4f:9206
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

verbose `lsusb` output:
```
Device: ID 1b4f:9206
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1b4f
  idProduct          0x9206
  bcdDevice            1.00
  iManufacturer           1 SparkFun
  iProduct                2 SparkFun Pro Micro
  iSerial                 3 HIDFED
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          125
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      0
      iInterface              0
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x01
[... 145 lines cut ...]
KeyboardConverter.handleKey: 0 --> 0

suniversal: 0 down

KeyboardConverter.handleKey: 0 --> 0

suniversal: 0 down

KeyboardConverter.handleKey: 0 --> 0
```

This was for a single key press of one single key. Which key does not matter, the output is always the same.

Here is some more detail on how the Arduino behaves when `DEBUG` is enabled:
* Hitting any key does nothing
* When `/dev/ttyACM0` gets opened (`cat /dev/ttyACM0`), the Arduino blinks a yellow LED a single time. Same when `/dev/ttyACM0` gets closed again
* When a key gets pressed while `/dev/ttyACM0` is getting read from (`cat /dev/ttyACM0`) another LED flashes bright green on each key press
* When CAPS lock is pressed on my laptop keyboard, both LEDs (yellow and bright green) flash

Here are some pictures of:
* [the wiring](https://oc.honet.ch/index.php/s/kSNDRXcckLBmeja#//IMG_20190329_030215.jpg)
* [the connector](https://oc.honet.ch/index.php/s/kSNDRXcckLBmeja#//IMG_20190328_191906.jpg)
* [the connector and our wiring](https://oc.honet.ch/index.php/s/kSNDRXcckLBmeja#//IMG_20190329_030158.jpg)
* [the keyboard's mainboard](https://oc.honet.ch/index.php/s/kSNDRXcckLBmeja#//IMG_20190328_191914.jpg)
* (just for fun) [our development setup](https://oc.honet.ch/index.php/s/kSNDRXcckLBmeja#//IMG_20190329_030132.jpg)

xelalexv, 2019-03-30:
Thanks for your (very detailed, thumbs up) report! But first off, congratulations on a nice keyboard! With MX switches, that's sweet :-)

Regarding your problem: From what I could find, based on the [FCC ID](https://fccid.io/K6XS5100) this is actually a keyboard made by Integrix. And [this message](http://www.silug.org/lists/silug-discuss/200104/msg00031.html) in a fairly old mail thread suggest that they were not actually compatible with original Sun keyboards. But, we may still get this beauty to work.

From what you describe, your overall setup seems correct. The Arduino receives something from the keyboard upon a key press, but the value is always 0, which does not get forwarded to your PC, since 0 is not a key code. I suspect that the keyboard is either using a different baud rate, or is not using inverted serial signals as the Sun keyboards do. You could try playing around with these two parameters in `suniversal.ino`: signal inversion is the boolean in line 71, baud rate is set in line 95. Keep me posted!

xelalexv, 2019-03-31:
One more thought. In the last of your pictures I see that you have an oscilloscope. Maybe you could capture the serial signal coming from the keyboard for a single key press. We could then try to deduce the serial settings from that.

NiklausHofer, 2019-03-31:
Thank you so much for offering to help with this!

I had not realized that this keyboard was not actually manufactured by SUN. But now that you mention it, the hints have always been there. The name 'Integrix' is even molded into the case (on the inside, so without opening the keyboard one doesn't get to see it).

The Oscilloscope in the picture was used by a friend to help me debug the problem. He has suspected that something funny is going on with the signal since pretty early in the process. I didn't quite understand and don't remember everything he told me, but I will meet him again on Tuesday and will certainly keep you posted.

According to his analysis, the serial signal is inverted. The baud rate, he says, should be 10500. I don't understand how he has come to this conclusion, but I will ask him to explain again and also send along an image of the Oscilloscope's screen.

We have toyed around with the Baud rate and inverted signal variables already. In most cases, after setting a different baud rate, the serial interface would stop outputting any debug information. We got output on 600, but it looked exactly the same as in my first posting with the final code always being <code>0</code>.

Changing the Boolean for the inverted signal had the expected effect - instead of <code>0</code> I get <code>7f</code>, with everything else being unchanged. I cannot copy & paste any output right now, nor can I test if keystrokes are sent to the computer using <code>xev</code> since I left the keyboard at the local hackerspace.

After seeing you message last night I tried some more combinations of the inverted signal and baud rate, but without success.

Unfortunately the SoftwareSerial library does not support a 10500 baud rate according to [the documentation](https://www.arduino.cc/en/Reference/SoftwareSerialBegin). I hope we don't end up having to patch the library. Before attempting to do so, I will make sure to send you the requested information from the Oscilloscope and wait for your verdict.

NiklausHofer, 2019-04-03:
Here's the Osciloscope's output for the key with letter 's': https://oc.honet.ch/index.php/s/kSNDRXcckLBmeja#//IMG_20190402_195012.jpg

xelalexv, 2019-04-04:
OK, this may get us somewhere. Here's my analysis so far.

I tried mapping the scan code for the `s` key to the bit pattern from the oscilloscope, but it didn't really make sense, until I realized that the little rascal may not actually be using SUN scan codes, but IBM ones ;-) With that, it seems to make sense:

scan codes for `s`:
```
SUN: 0x4E, MSB: 01001110, LSB: 01110010
IBM: 0x1B, MSB: 00011011, LSB: 11011000
```
notation:
`-` = idle
`{` = start bit
`}` = stop bit
```
position	a | b | c | d | e | f | g | h | i | k | l | m
   level	H | L | H | H | L | H | H | L | L | L | H | H
 logical	- | { | 1 | 1 | 0 | 1 | 1 | 0 | 0 | 0 | } | -
```
If I interpret the screen shot correctly, the oscilloscope was set to 80us per division, which gives us a bit more than 100us per bit, and that would [translate to 9,600 bits/s](https://en.wikipedia.org/wiki/Serial_port#Settings). I think the other settings then would be 8 data bits, no parity, 1 stop bit (or 2, we can't really tell). Data bits are sent in LSB order, i.e. standard. Non-inverted logic is used. I'm not totally sure about positions `h` through `k`. That part of the sequence is too short for just two bits, but from the looks of it also not really long enough for three bits. Well, maybe just try with this setting, i.e.:
```c++
// for communication with the SUN keyboard
SoftwareSerial sun(PIN_RX, PIN_TX, false);
...
void setup() {
...
    sun.begin(9600);
...
}
```
If that works, the next step would be to create an IBM-to-USB table for proper key translation.

NiklausHofer, 2019-04-07:
We got most of the keyboard functionality working last night. I am afraid though, that the code does not make a whole lot of sense.

  https://github.com/NiklausHofer/suniversal/tree/integrix

We tried 9600 Baud before but were not lucky. We would get different keys identified as the same. Currently we have 10500 working. For all example keys we have tried, manually reading the waves from the Oscilloscope produces the same number as does the Arduino. So it's mostly fine.

Unfortunately the scan codes do not appear to make any sort of sense. I ended up pressing each key, reading the debug output and writing the code down. It works, but the codes are seemingly random. In an attempt to make some more sense, I have tried inverting them and even reading them back to front, but without any luck.

```
Q    0x15    0001 0101
W    0x1D    0001 1101
E    0x24    0010 0100
R    0x2D    0010 1101
T    0x2C    0010 1100
Y    0x35    0011 0101
```

We have also had to handle the release key sequence differently. Upon pressing a key, the keyboard sends the code. Upon releasing the key, the keyboard sends <code>0xf0</code> followed by the key code. That bit was easy though.

The really bad thing though are the special keys (INSERT, DELETE, ... and the arrow keys) which behave completely different from all other keys. They send a sequence like this:

`0xe0 0x12 0xeo 0x70` with the first three bytes (`0xe0 0x12 0xeo`) being the same for all these keys and only the last one changing from key to key. Unfortunately the last byte collides with the code for other keys on the keyboard, so it is not an option to simply drop the three proceeding bytes. To make this a bit more clear: The above code (`0xe0 0x12 0xeo 0x70`) identifies the INSERT key, but the last byte (`0x70`) on it's own identifies the key `0` on the number pad. I guess that makes some kind of sense because `0` on the number pad is also INSERT but it is very annoying none-the less. I have written code to handle these keys as well, but I will want to have another look with the Oscilloscope, especially since releasing these keys works only about 90% of the time.

xelalexv, 2019-04-07:
I'm glad to hear you made some progress!

Your scan codes are actually spot on, and totally make sense (see [here](https://www.seasip.info/VintagePC/ibm_1397000.html), it's set 2!). So the keyboard is using IBM scan codes after all. Also your other observations such as the make & break behavior, are consistent with an IBM keyboard ([check this](https://www.seasip.info/VintagePC/ibm_1397000.html)).

About the electrical interface, 10,500 bps is very odd in deed. Maybe what we have here is actually a PS/2 interface? 10,500 Hz clock rate would be right within the specs (have a look [here](http://www.burtonsys.com/ps2_chapweske.htm)). To make sure, check with your oscilloscope whether any of the other wires has a clock signal that is in sync with the data line. If this really is PS/2, then all you need is a USB adapter for that, and I'm sure there are drivers on Linux that can work with the keyboard.

xelalexv, 2019-04-29:
@NiklausHofer Have you had any further progress with this? If you got it working, whether with suniversal or something else, it would be great to hear about it, so I can reference your solution for others in the documentation.

NiklausHofer, 2019-05-07:
@xelalexv The code in https://github.com/NiklausHofer/suniversal works mostly. I have not added all the keys as of yet, only those that I commonly use. Also the LEDs are not yet working.

xelalexv, 2020-05-29:
Closing due to inactivity. Please re-open if there are any new insights into this. In particular, I'd be interested to know whether the interface is actually PS/2, as I suspected further up.

closed 2020-05-29 by xelalexv