xelalexv/suniversal #5: Compose key flicks on and immediately off again in COMPOSE_MODE

wyatt8740, 2019-08-11:
I set `COMPOSE_MODE = true` in my build, and while the key does send to the computer, the LED flicks on for a fraction of a second before turning off again, even before I have a chance to release the compose key!
I am assuming this is not the desired behavior, since it is documented to stay on for two key presses or until compose is pressed again.

xelalexv, 2019-08-11:
What is your host OS, and have you activated the compose key there? I'm currently on vacation and can't check myself. That would have to wait until next week.

xelalexv, 2019-08-11:
Oh, and could you turn on debugging and attach the serial output here?

wyatt8740, 2019-08-12:
Debian sid. And will do shortly - buying another uC board in the mail though since I just snapped the mini USB port off and lifted traces attempting to fix it -_-

It's a Unix layout US type 5C, by the way.

xelalexv, 2019-08-12:
From the logs it looks like the host OS is turning off the compose LED right away (the `LED state changed: 3 --> 1` message at the top). I faintly remember that on my Ubuntu box, the compose LED would never light up if I let the host OS control it, because it would always send state `0` for that LED. That's why there is some logic in `suniversal.ino` to handle that on the keyboard side. Seems like a bug though that in `updateLEDs` the state from the host gets applied to the compose LED anyways.

I created a branch `issue-#5`. The changes let you configure whether you either want the host OS to control the LED, or suniversal (`COMPOSE_LED_HOST_CONTROLLED` in `config.h`). Give it try with both options and let me know.

wyatt8740, 2019-08-13:
Ok, so with it set to 'false' the led comes on for two key strokes. It doesn't even blink on in 'true' mode.

It should also be stated though that not all compose key combos are two characters in length, so it might be better to listen for the OS and toggle states when detected.

Examples `include <compose> --.` for an en dash, `<compose> ---` for an em dash.

xelalexv, 2019-08-14:
Your findings confirm my flaky memory ;-) i.e. what you see is what I expected. With `COMPOSE_LED_HOST_CONTROLLED` set to `true`, the host has full control over the compose LED. Only, it never ever turns it on. Setting to `false` lets suniversal control the LED, with the drawback that it stays on for a fixed length of two key strokes, which is ok for an approximation of the feature.

I agree that for fully correct control of the compose LED, the OS should take charge. Problem is that Linux just doesn't seem to care. I think the HID descriptor in `usb_keyboard.cpp` correctly defines the presence of the compose LED, but maybe you can double check with `sudo cat /sys/kernel/debug/hid/<interface of the keyboard>/rdesc`. The output should contain something like this:

```
LED.NumLock ---> LED.NumLock
LED.CapsLock ---> LED.CapsLock
LED.ScrollLock ---> LED.ScrollLock
LED.Compose ---> LED.Compose
LED.Kana ---> LED.Kana
```

I also added more debug output regarding the LED state changes, so maybe you could attach that here too, for the case when `COMPOSE_LED_HOST_CONTROLLED` is set to `true`.

xelalexv, 2019-08-28:
I would propose to merge the changes from the issue branch into `master`, since having the additional config option makes sense. Did you get anywhere with host controlled LED?

xelalexv, 2020-05-30:
All commits have been merged to `master`. Conclusion is that Linux is simply not controlling the Compose LED. If anyone can provide some info on how to change that, please reopen.

closed 2020-05-30 by xelalexv

wyatt8740, 2020-06-04:
I forgot to follow up. I'll try to do that when I next can. Sorry for the lack of reply earlier.

jabuzz, 2020-08-29:
I would note that on a Sun workstation (much the same way as on a DEC LK201/VT220 combination) that there where only ever two key compose sequences. The three key ones are all new fangled Linux inventions.

wyatt8740, 2020-09-01:
Well, that'd be a fair point. Unfortunately I am too young (and too poor) to have a Sun workstation (or VT220).

Isn't the compose LED still controlled by the workstation itself, even on the original machines?

xelalexv, 2020-09-01:
I'd say that the compose LED is always controlled by the host, because that's where the processing of the sequence happens and only there do we know when the sequence is complete.

Regarding the original fixed length compose sequences, that's an interesting historical fact. Since _suniversal_ is concerned with making the _Sun_ keyboards work on present day machines, we'll have to consider the "new fangled" ones though :smile: