xelalexv/suniversal #6: when 3 keys held some combinations not working

rabfulton, 2019-10-02:
Firstly thank you for this wonderful project, love my arduino enabled type 5 keyboard.

I have discovered an issue on linux(arch) whereby if you press and hold in sequence: 
up arrow; right arrow; space bar, 
the space bar key event is not sent. it can also be produced by various other combinations such as: 
up arrow; v; right arrow, 
right arrow is not sent, but they all seem to have the right arrow key in common (at least as far as I have found).

xelalexv, 2019-10-03:
Thanks for the feedback! I confirmed this problem on my machine. Could you run suniversal in debug mode and attach a log here? I currently don't have the development environment fully set up, so that would help.

rabfulton, 2019-10-04:
Debug info for up-right-space:

KeyboardConverter.handleMacro: 2a, 0 --> not a macro
KeyReport.handleModifier: 0, 0 --> not a modifier
KeyReport.removeKey: 2a --> removing from slot 0
KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]

suniversal: all released
KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]

suniversal: 14 down
KeyboardConverter.handleKey: 14 --> 52
KeyboardConverter.handleMacro: 52, 1 --> not a macro
KeyReport.handleModifier: 0, 1 --> not a modifier
KeyReport.addKey: 52 --> adding at slot 0
KeyReport.send: modifiers=0, keys=[ 52 0 0 0 0 0 ]

suniversal: 1c down
KeyboardConverter.handleKey: 1c --> 4f
KeyboardConverter.handleMacro: 4f, 1 --> not a macro
KeyReport.handleModifier: 0, 1 --> not a modifier
KeyReport.addKey: 4f --> adding at slot 1
KeyReport.send: modifiers=0, keys=[ 52 4f 0 0 0 0 ]

suniversal: 1c up
KeyboardConverter.handleKey: 1c --> 4f
KeyboardConverter.handleMacro: 4f, 0 --> not a macro
KeyReport.handleModifier: 0, 0 --> not a modifier
KeyReport.removeKey: 4f --> removing from slot 1
KeyReport.send: modifiers=0, keys=[ 52 0 0 0 0 0 ]

suniversal: 14 up
KeyboardConverter.handleKey: 14 --> 52
KeyboardConverter.handleMacro: 52, 0 --> not a macro
KeyReport.handleModifier: 0, 0 --> not a modifier
KeyReport.removeKey: 52 --> removing from slot 0
KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]

suniversal: all released
KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]

rabfulton, 2019-10-04:
A quick examination shows that while up-right are held all of the lower letters z,x,c,v,b,n,m, comma, \ and space are not working, other keys are fine.

xelalexv, 2019-10-04:
Judging from the log, it seems the _SUN_ keyboard doesn't send the third key. For every key that's received from the keyboard via the serial port, a log line such as `suniversal: 14 up` or `suniversal: all released` will show. From a quick glance over the code (`loop()` and `handleKey(uint8_t)` in `suniversal.ino`), I don't see anything that would prevent a key from being logged once it is received. I'll double check in my env.

rabfulton, 2019-10-04:
Feels like it could be some kind of hardware issue. I checked using caps lock on and it does not make any difference. Would be interesting if someone could try this on a sun computer without any converter. Quite annoying as such combinations a quite common in games.

xelalexv, 2019-10-04:
I confirmed in my env by adding extra debugging info. When the third key in the mentioned three key combinations is pressed, nothing is sent by the _SUN_ keyboard. So this really looks like a h/w limitation. Would be great though to confirm this with an actual _SUN_ machine. Sadly, my SPARC Station 2 won't boot any more... :-(

closed 2019-10-05 by rabfulton