xelalexv/suniversal #8: Power key and USB power budgets

jabuzz, 2020-08-29:
There is a way to potentially stay within the USB standby power budget and have the power key function. This is assuming the Arduino board can be powered from the USB standby power (though technically you don't even need the Arduino powered up I think though that would get more complicated).

The trick is to understand how the power key works on a Type 5 keyboard. If you dismantle one and trace the lines on the membranes you will see that the power key has it's own dedicated trace connected to the last pin on each membrane connector (pins 26 and 9). I would at this point note that some documentation on the pin out for a Sun keyboard will have pin 7 marked as "keyboard power".

Now if you trace the circuit on the keyboard PCB you will see that pin 7 on the mini DIN is connected to the cathode of a diode (D12 on the PCB's I have for a Type 5c keyboard; I have several), with the anode being connected to pin 26 on the wide membrane connector with pin 9 on the narrow one being connected to ground aka pins 1/2. Put a multimeter over pins 7 and 1/2 in diode test mode and you can see it go from open loop to 0.6V when you press the power button.

In addition pins 3/8 aka 5V on the mini DIN are connected via a 33kΩ (R8) resistor to pin 20 on the MC68HC, and also to the cathod of a diode (D13)  the anode of which is connected to pin 26 on the membrane connector, and hence to the anode of the other diode (D12). Once the keyboard is powered up pressing the power key will pull pin 20 low.

As such when you are on standby USB power you can depower the main keyboard and just watch for pin7/ KBD PWR to be pulled low. At that point you can turn the computer on and when the USB port is properly powered up turn on the keyboard. I would suggest using an n-channel MOSFET driven by a pin on the Arduino to control the power to the keyboard.

I can't remember now if the power key on a real Sun workstation does anything once the machine is powered up and unfortunately my SPARCstation 5 is dead (need to recap the PSU) and I would imagine the NVRAM is dead too.

Question is would you be willing to take patches for that sort of functionality (given it would require hardware modifications) or should I create a fork?

xelalexv, 2020-08-30:
I'd love a working power key :-) We could keep the additional h/w optional, just like the inverter for the mouse. Maybe you could do a few experiments first for USB wake up with a bare Arduino Micro. I haven't really been able to get this working, haven't tried that hard though. At the office I have a Type 5 connected through the USB hub in the monitor, which is powered even if the monitor is in standby. So there, the keyboard is fully on all the time, but sending the USB power key code has no effect. So I think we should also establish what exactly is required on the host side for USB wake up.

jabuzz, 2020-08-30:
The background to all this is that what is my now RYZENstation ITX project is out of hibernation after ~15 years. The idea is to put a PC inside an IPX case in "stealth" mode, aka all the original ports function as they would on a real IPX. As such I will probably be doing it differently in that I will be using a small 1W PSU for standby power which will be used to power the Arduino and I will be using the motherboard power button and power LED headers to switch the machine on and determine whether it's powered up. I also need to power a relay for the pass through power and switch on the main PSU for the system.

However I would like to make it work if possible properly for everyone else. There is something pleasing about a perfect converter. Some research today suggests that the Arduino should be below the 50mA USB standby current at 45mA. Though if you are willing to remove the power LED you can save 5mA, which should then leave one with more than enough. I need to find a USB cable I can vandalise to measure the power accurately (I doubt the cheap USB power meters are sufficiently accurate) just to make sure.

I will need to dig into the USB specification to work out if you can tell what power state you are in. That is you need to know that are not on standby power before you power up the keyboard, otherwise it's a bust. Presumably it is possible but as they say presumption is the mother of all screw ups.

I have a schematic of the keyboard power functionality that I could upload if you want?

jabuzz, 2020-08-31:
Done some research and I have ordered up a couple of Infineon IRFB7430PBF MOSFET's from RS (stock no 776-9172). Specification wise it looks a bit over the top, but the potential issue is having a low maximum drain source resistance. If that's too high then we will get too much voltage drop on the supply and the keyboard won't work. Your typical low power TO-92 package device will have an on resistance of around 0.6Ω where as the IRFB7430PBF has an on resistance of just 1.3 mΩ. Of course that means it comes in a TO-220 package and can handle 195A when heatsinked. However it's only £2.8 and bizarrely €2.75 from RS Germany. I picked this one simply because it was the lowest through hole on resistance n-channel MOSFET that I could order singly.

Hardware wise my thinking is we can just stick this in the power line then, attach the gate to one of the IO pins. My next idea is then to put a jumper on one of the other IO pins to pull it low. We can then have the software look for that pin being low and if it is assume that we have keyboard/mouse power control in place and that way you only need the one software image.

As a side note you could wire up a dip switch or jumpers, so that things like macros and scroll wheel emulation can be selected at run time rather than having to recompile if you want to change options.

xelalexv, 2020-09-01:
I wouldn't worry that much about the voltage drop. The Type 5c is rated at 150mA max, so 0.6Ω would give a max drop of just about 0.1V. I'm sure the keyboard will work quite happily with as little as 4.5V, so you could even use a 2N2222 (as per data sheet, with Ib @ 15mA and Ic @ 150mA, Vce should be 300~400mV). Alternatively, an old school miniature relay would work too.

jabuzz, 2020-09-01:
I would never use a bipolar transistor in this scenario, you are always better off with a MOSFET. I was suffering from a brain fart on Monday, something like a IRLZ34NPBF (RS part 541-1247) would be a more sensible choice and only £1. Still in a TO-220 package but with a maximum drain source resistance of 35mΩ, then with the 150mA for the Type5c keyboard and 60mA for a Type 6 cross bow mouse (some inspection of pictures on eBay of Sun mice being sold shows this is the highest draw current of the lot) then the voltage drop would be just 0.00735V. Nothing to worry about. So a much lower voltage drop than you will ever get with a bioplar transistor. You could use something like a  ZVN4306A (RS part 841-299) which comes in an E-line package (very similar to a TO-92). However this has a maximum drain source resistance of 300mΩ, giving a voltage drop of 0.063V costs more and can only be ordered (from RS at least) in multiples of five. If I where doing a production run of thousands I would probably use this, in the meantime a TO220 (or similar) packaged MOSFET is a better choice.

The other advantage of the MOSFET is that you don't need any additional components. Being voltage controlled rather than current controlled you don't need a resistor on the base so doing a dead leg construction is much easier IMHO.

xelalexv, 2020-09-01:
MOSFETs certainly have their advantages over bipolar transistors, but in a hobbyist setting, I'd say whatever is available and works is fine. I'd have no problem with using a bipolar transistor for this particular application, if that's what I have in my parts box and don't need to order. Keyboard and mouse are going to be fine with just 4.5V. If I already have a MOSFET, I'd take that. I'd also prefer the additional resistor over a part in a TO-220 package, but that's just me.

xelalexv, 2020-09-02:
Regarding your side note about the DIP switches for runtime config further up, I like that idea. When I started _suniversal_, I was was focused on getting it working with minimal h/w. I also wanted the converter to be in the keyboard case, but not make any modifications to it, so this never crossed my mind. I'll make it a separate enhancement issue.