xelalexv/suniversal #9: Compose Key under Windows

jabuzz, 2020-08-29:
I have an idea about the compose key :-)

First IMHO the compose key should never be married to the Windows menu key, that is heresy. IMHO it would be better paired with the "Props" key in the fun cluster. That would be a more natural pairing (well at least for a die hard like myself).

Now to the genesis of my idea, which is to make the Compose key work under Windows without needing to install anything in Windows. It should work in Linux too as well without needing to enable a compose key.

Basically it works like this. You hit the compose key and we light up the compose LED. Then we watch for a compose sequence (exiting and turning the LED off if a valid one is not entered). If we get a valid compose sequence we check the status of Num lock, toggle if necessary (but don't change the LED) and then send an Alt code that would correspond to the compose sequence, turn off the compose LED and toggle Num lock as required.

We can match the Alt codes to relevant code pages (I guess mainly 437 and 850) based on the keyboard layout as reported by the Sun keyboard itself

Might not be as flexible as having the OS do it, but then again the OS can't do it on Windows without installing third party software.

What do you think and would you be willing to accept patches for this?

xelalexv, 2020-09-01:
The meaning of the USB HID code `USB_COMPOSE` (`0x65`) for the compose key is _Keyboard Application_, and that's apparently mapped by Windows to the menu key (the box with the three dashes inside). The current mapping of the compose key under Windows is the same as that of a _Sun_ Type 7 (just double checked). The Type 7 was actually my standard  when defining the mappings. For a different mapping, you can edit `sun_to_usb.h`.

jabuzz, 2020-09-02:
Sure I understand that. I am proposing something more than a mapping of the key to something different.

What I am getting at is that under Windows a compose key requires extra software to be useful, but we could make it useful without requiring any additional software by emulating the compose key in the "keyboard". That is hitting the compose key does not send a code to the computer but starts a compose sequence that the Arduino interprets that if valid quickly sends an ALT code instead to generate that character. 

That is you hit compose ^ e and then the Ardunio Sees the compose key depressed and lights the compose LED. It then looks to see if you type a valid compose sequence that it knows about. It sees that ^ e is a valid compose sequence so quickly toggles Num lock on if needed, sends ALT 1 3 6 before toggling Num lock off if needed, and then turning off the Compose LED. Note even if we change the Num lock status on the host we don't change the LED on the keyboard so it looks to the user like it it was not interfered with.

This would be a bit similar to how it was done originally on a DEC VT220 terminal, which never sends any information from compose key on the LK201 keyboard to the host, just the finished character from a valid compose sequence.

Obviously this would be optional because it won't work under Linux/MacOS which don't  do ALT codes, and besides Linux understands the compose key out the box. There is presumably something similar on Mac's but I have no idea.

xelalexv, 2020-09-02:
No worries, I got the idea right from the start, and I think it is an excellent fallback solution, at least for Windows. I can check on Mac for something similar to ALT sequences. It's just that yesterday I only had time to double check & reply to the first item in your post regarding the mapping of the compose key under Windows. Sorry about that half way reply.