xelalexv/suniversal #10: change config settings with DIP switches or jumpers

xelalexv, 2020-09-02:
As suggested [here](https://github.com/xelalexv/suniversal/issues/8#issuecomment-683665496) by @jabuzz, we could use DIP switches or jumpers for defining config settings, so that we don't have to upload _suniversal_ to the _Arduino_ each time we want to make a config change. A minor drawback of this is that to make the switches or jumpers accessible, you either need to put the converter into a separate case or make modifications to the keyboard case. This feature should therefore be optional.

mikaelsundin, 2020-09-11:
Maybe implement this with pull-up on spare gpio and have all high as default mode as behave as previous fw.

xelalexv, 2020-09-12:
That would be the right approach. On an _Arduino Pro Micro_, we still have 14 GPIOs left, so a lot of space for settings. All GPIOs support internal pull-up resistor, just needs to be configured. So we would really just need DIP switches/jumpers to ground.

However, we would need one additional config setting, that defines whether DIP switches/jumpers are fitted and take precedence over the software config settings. Otherwise we could not distinguish between a DIP switch/jumper saying HIGH vs. no DIP switch/jumper present.

prosper00, 2021-03-18:
I was actually considering something similar - a jumper setting to enable/disable the usb serial port, so that I don't accidentally flash my keyboard inadvertently when I'm working on another arduino.

I had considered just erasing the bootloader (and flashing suniversal with a programmer like the USBisp), but I do like to tinker and play with my keymaps now and then.... I suppose I could just break out the ISP header somewhere too