xelalexv/suniversal #12: Sun type 3 connection

hellyhot, 2020-11-25:
Hi, 

not sure how I can contribute to the github but I have a sun type 3 and since you asked for feedback on using this code with it I guess I would share my experience 

Here's how the connection to the arduino looks like : 
[image]

I can say much else for now since I broke the micro-usb on my arduino trying to secure it inside the keyboard... but during initial testing all keys looked to work

xelalexv, 2020-11-26:
Thanks for the feedback, that's great news! I'll add that to the readme. Re the Arduino mishap, you're not alone - the exact same thing happened to me, knocked the connector off while looking for a good place to put it inside the case. Maybe I should also add a warning to that effect to the readme... :rofl:

xelalexv, 2020-11-26:
BTW, may I use your image in the readme?

hellyhot, 2020-11-27:
Of course you can use it ! Let me know if you need more test on the type 3

xelalexv, 2020-11-27:
No specific tests, just let me know if something doesn't work. Now that I know it works for Type 3, I really need to get hold of one...

hellyhot, 2020-11-27:
Hey, writing this from my type 3 now actually.

Received the new arduino and didn't broke that one up so that's good.

Right now there's some issue that I can't seems to be able to fix : 

    1) cannot find how to disable keyboard beeping on every key press 
    2) fun cluster isn't working but it may be broken on my keyboard since keys are not beeping like the others
    3) how should I proceed to remap keys to my preference ? modifiying sun_to_usb or usb_codes ?

Again thanks for your help on this. This is my first Arduino project... way difference that web dev ahah

xelalexv, 2020-11-28:
Re 1) Is that really a beep on each key press, or more like a click? The SUN keyboards do have a key click function, which according to section 3.2.2 of the SPARC keyboard spec is a 5ms beep at approx. 2kHz. For the Type 5 keyboards, this key click is off after power on. Maybe it's on for the Type 3. I created branch `gh-12`, in which the key click is explicitly turned off during keyboard reset. Maybe you can try that.

re 2) Could you turn on debug messages, try again and post the output here? Could maybe also help with 1), if the patch in `gh-12` does not help.

re 3) `sun_to_usb.h` would be the right place.

And yes, Arduino stuff is quite different. I work with Kubernetes in my day job, so pretty much on the other end of the spectrum. Tinkering with micro-controllers makes for a nice change :smile:

hellyhot, 2020-11-28:
I tried with the patch from gh-12 , also tried CMD_BELL_OFF in case without luck. 

I'll play around with it more tomorrow

xelalexv, 2020-11-28:
Debug output should help us. Another thought, are you sure you got the TX line to the keyboard right? In [this project here](http://blog.daveastels.com.s3-website-us-west-2.amazonaws.com/2014/12/27/type-3-keyboard.html), I can see that pin 5 on connector J4 is used instead of pin 4. Maybe it's worth double checking. The pin out of the original DB15 connector of the keyboard would be [here](http://www.sunhelp.org/faq/sunref5.html).

hellyhot, 2020-12-01:
Could it be the arduino IDE that I use to compile/upload ? it seems to throw quite a lot of error before finishing uploading the code

hellyhot, 2020-12-01:
I think that was the issue. using Arduino IDE 1.8 at home wasn't working right somehow , recompiled the code and uploaded via arduino online ide worked at work.

thanks for all your help !

xelalexv, 2020-12-01:
So is the functionality ok now, i.e. keys working as expected and beep gone?

hellyhot, 2020-12-01:
beep is gone 
fun cluster isn't working but I get no key registration on the debugging , I'm pretty sure it's just my board that is broken

xelalexv, 2020-12-02:
Looks like it, which is sad. If the keyboard were sending anything while pressing a key in the fun cluster, we would see that in the debug log. Maybe it's just a single trace on the PCB that needs to get fixed.

So we can confirm that the connection to J4 on the keyboard is correct as per your original photo?

Meanwhile you can keep this issue open and do some more testing. When no other problems arise and you're satisfied, just close.

xelalexv, 2020-12-16:
Closing for now, as there doesn't seem to be any remaining issue that would require action. Please reopen otherwise.

closed 2020-12-16 by xelalexv