xelalexv/suniversal #13: (my personal) fix for linux issue on writing code to arduino

newm-newm, 2020-12-02:
Initially, writing to the arduino on Linux is hit or miss. Sometimes it will work, most times it wont.

What I've personally found is to run the arduino ide as root and to issue the command: `sudo systemctl stop ModemManager.service`. With doing both of these, I haven't had an issue uploading any code to a board. Not aware if it's a known issue or not. Just some helpful info. I just packaged this into a quick bash script to run that command and then run arduino as root.

xelalexv, 2020-12-02:
Thanks for the tip!

If you still need the `modemmanager` package on your system, then stopping `ModemManager` service before running the IDE is a good idea. Otherwise, as pointed out in the readme under _Development_, you can also completely remove the `modemmanager` package and wouldn't have to worry any longer.

Regarding running the IDE as root, I'd personally be a bit reluctant to do so. When you're running it as non-root, is the upload occasionally not working, or not at all? If not at all, then you may have to add your non-root user to the `dialout` group. Otherwise it will not have access to the USB serial port.

newm-newm, 2020-12-04:
> Regarding running the IDE as root, I'd personally be a bit reluctant to do so. When you're running it as non-root, is the upload occasionally not working, or not at all? If not at all, then you may [...]

It's intermittent unfortunately. I've also already added my user to the `dialout` group.

xelalexv, 2020-12-05:
OK. I'll add this to the Readme.