xelalexv/suniversal #14: LED's (caps, scroll, num, etc) not working

prosper00, 2021-02-01:
I built this up (using a Leonardo, which is essentially identical to the pro micro). Seeing some strange behaviour with the LED's. When the USB is first connected, the LED's do flash very briefly and the beeper sounds. Pressing caps on the keyboard DOES work (it makes the characters come in as caps), and the light on my other keyboard toggles correctly. However, the lights on the sun board don't change. I see the RX/TX lights on the arduino indicate activity, and if I connect a scope to the RX pin of the keyboard PCB, I do see activity coming in (in to the keyboard from the Arduino).

Otherwise, kepyresses from the Sun DO correctly come into to the PC

xelalexv, 2021-02-01:
Could you run suniversal in debug mode and post the log here?

prosper00, 2021-02-01:
```
could not determine keyboard layout, defaulting to US suniversal: USB LED state changed: 0 --> 1 suniversal: SUN LED state changed: 0 --> 1 suniversal: raw key 94 suniversal: 14 up KeyboardConverter.handleKey: 14 --> 52 KeyboardConverter.handleMacro: 52, 0 --> not a macro KeyReport.handleModifier: 0, 0 --> not a modifier KeyReport.removeKey: 52 --> not found  suniversal: raw key 7f suniversal: all released KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 14 suniversal: 14 down KeyboardConverter.handleKey: 14 --> 52 KeyboardConverter.handleMacro: 52, 1 --> not a macro KeyReport.handleModifier: 0, 1 --> not a modifier KeyReport.addKey: 52 --> adding at slot 0 KeyReport.send: modifiers=0, keys=[ 52 0 0 0 0 0 ]  suniversal: raw key 94 suniversal: 14 up KeyboardConverter.handleKey: 14 --> 52 KeyboardConverter.handleMacro: 52, 0 --> not a macro KeyReport.handleModifier: 0, 0 --> not a modifier KeyReport.removeKey: 52 --> removing from slot 0 KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 7f suniversal: all released KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 1b suniversal: 1b down KeyboardConverter.handleKey: 1b --> 51 KeyboardConverter.handleMacro: 51, 1 --> not a macro KeyReport.handleModifier: 0, 1 --> not a modifier KeyReport.addKey: 51 --> adding at slot 0 KeyReport.send: modifiers=0, keys=[ 51 0 0 0 0 0 ]  suniversal: raw key 9b suniversal: 1b up KeyboardConverter.handleKey: 1b --> 51 KeyboardConverter.handleMacro: 51, 0 --> not a macro KeyReport.handleModifier: 0, 0 --> not a modifier KeyReport.removeKey: 51 --> removing from slot 0 KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 7f suniversal: all released KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 1b suniversal: 1b down KeyboardConverter.handleKey: 1b --> 51 KeyboardConverter.handleMacro: 51, 1 --> not a macro KeyReport.handleModifier: 0, 1 --> not a modifier KeyReport.addKey: 51 --> adding at slot 0 KeyReport.send: modifiers=0, keys=[ 51 0 0 0 0 0 ]  suniversal: raw key 9b suniversal: 1b up KeyboardConverter.handleKey: 1b --> 51 KeyboardConverter.handleMacro: 51, 0 --> not a macro KeyReport.handleModifier: 0, 0 --> not a modifier KeyReport.removeKey: 51 --> removing from slot 0 KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 7f suniversal: all released KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 14 suniversal: 14 down KeyboardConverter.handleKey: 14 --> 52 KeyboardConverter.handleMacro: 52, 1 --> not a macro KeyReport.handleModifier: 0, 1 --> not a modifier KeyReport.addKey: 52 --> adding at slot 0 KeyReport.send: modifiers=0, keys=[ 52 0 0 0 0 0 ]  suniversal: raw key 94 suniversal: 14 up KeyboardConverter.handleKey: 14 --> 52 KeyboardConverter.handleMacro: 52, 0 --> not a macro KeyReport.handleModifier: 0, 0 --> not a modifier KeyReport.removeKey: 52 --> removing from slot 0 KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 7f suniversal: all released KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 14 suniversal: 14 down KeyboardConverter.handleKey: 14 --> 52 KeyboardConverter.handleMacro: 52, 1 --> not a macro KeyReport.handleModifier: 0, 1 --> not a modifier KeyReport.addKey: 52 --> adding at slot 0 KeyReport.send: modifiers=0, keys=[ 52 0 0 0 0 0 ]  suniversal: raw key 94 suniversal: 14 up KeyboardConverter.handleKey: 14 --> 52 KeyboardConverter.handleMacro: 52, 0 --> not a macro KeyReport.handleModifier: 0, 0 --> not a modifier KeyReport.removeKey: 52 --> removing from slot 0 KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 7f suniversal: all released KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 14 suniversal: 14 down KeyboardConverter.handleKey: 14 --> 52 KeyboardConverter.handleMacro: 52, 1 --> not a macro KeyReport.handleModifier: 0, 1 --> not a modifier KeyReport.addKey: 52 --> adding at slot 0 KeyReport.send: modifiers=0, keys=[ 52 0 0 0 0 0 ]  suniversal: raw key 94 suniversal: 14 up KeyboardConverter.handleKey: 14 --> 52 KeyboardConverter.handleMacro: 52, 0 --> not a macro KeyReport.handleModifier: 0, 0 --> not a modifier KeyReport.removeKey: 52 --> removing from slot 0 KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 7f suniversal: all released KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 1b suniversal: 1b down KeyboardConverter.handleKey: 1b --> 51 KeyboardConverter.handleMacro: 51, 1 --> not a macro KeyReport.handleModifier: 0, 1 --> not a modifier KeyReport.addKey: 51 --> adding at slot 0 KeyReport.send: modifiers=0, keys=[ 51 0 0 0 0 0 ]  suniversal: raw key 9b suniversal: 1b up KeyboardConverter.handleKey: 1b --> 51 KeyboardConverter.handleMacro: 51, 0 --> not a macro KeyReport.handleModifier: 0, 0 --> not a modifier KeyReport.removeKey: 51 --> removing from slot 0 KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 7f suniversal: all released KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 1b suniversal: 1b down KeyboardConverter.handleKey: 1b --> 51 KeyboardConverter.handleMacro: 51, 1 --> not a macro KeyReport.handleModifier: 0, 1 --> not a modifier KeyReport.addKey: 51 --> adding at slot 0 KeyReport.send: modifiers=0, keys=[ 51 0 0 0 0 0 ]  suniversal: raw key 9b suniversal: 1b up KeyboardConverter.handleKey: 1b --> 51 KeyboardConverter.handleMacro: 51, 0 --> not a ma [... cut ...]  1 --> not a modifier KeyReport.addKey: 51 --> adding at slot 0 KeyReport.send: modifiers=0, keys=[ 51 0 0 0 0 0 ]  suniversal: raw key 9b suniversal: 1b up KeyboardConverter.handleKey: 1b --> 51 KeyboardConverter.handleMacro: 51, 0 --> not a macro KeyReport.handleModifier: 0, 0 --> not a modifier KeyReport.removeKey: 51 --> removing from slot 0 KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]  suniversal: raw key 7f suniversal: all released KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]
```

prosper00, 2021-02-01:
the last 8 lines are when I toggle caps from my USB keyboard

prosper00, 2021-02-01:
Here's just pushing and releasing caps and numlock a couple times:

> 
> 14:59:38.604 -> suniversal: raw key 77
> 14:59:38.604 -> suniversal: 77 down
> 14:59:38.604 -> KeyboardConverter.handleKey: 77 --> 39
> 14:59:38.604 -> KeyboardConverter.handleMacro: 39, 1 --> not a macro
> 14:59:38.604 -> KeyReport.handleModifier: 0, 1 --> not a modifier
> 14:59:38.604 -> KeyReport.addKey: 39 --> adding at slot 0
> 14:59:38.604 -> KeyReport.send: modifiers=0, keys=[ 39 0 0 0 0 0 ]
> 14:59:38.604 -> 
> 14:59:38.604 -> suniversal: USB LED state changed: 1 --> 3
> 14:59:38.604 -> suniversal: SUN LED state changed: 1 --> 9
> 14:59:38.670 -> suniversal: raw key f7
> 14:59:38.670 -> suniversal: 77 up
> 14:59:38.670 -> KeyboardConverter.handleKey: 77 --> 39
> 14:59:38.670 -> KeyboardConverter.handleMacro: 39, 0 --> not a macro
> 14:59:38.670 -> KeyReport.handleModifier: 0, 0 --> not a modifier
> 14:59:38.670 -> KeyReport.removeKey: 39 --> removing from slot 0
> 14:59:38.670 -> KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]
> 14:59:38.670 -> 
> 14:59:38.670 -> suniversal: raw key 7f
> 14:59:38.670 -> suniversal: all released
> 14:59:38.670 -> KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]
> 14:59:38.670 -> 
> 14:59:39.298 -> suniversal: raw key 77
> 14:59:39.298 -> suniversal: 77 down
> 14:59:39.298 -> KeyboardConverter.handleKey: 77 --> 39
> 14:59:39.298 -> KeyboardConverter.handleMacro: 39, 1 --> not a macro
> 14:59:39.298 -> KeyReport.handleModifier: 0, 1 --> not a modifier
> 14:59:39.298 -> KeyReport.addKey: 39 --> adding at slot 0
> 14:59:39.298 -> KeyReport.send: modifiers=0, keys=[ 39 0 0 0 0 0 ]
> 14:59:39.298 -> 
> 14:59:39.397 -> suniversal: raw key f7
> 14:59:39.397 -> suniversal: 77 up
> 14:59:39.397 -> KeyboardConverter.handleKey: 77 --> 39
> 14:59:39.397 -> KeyboardConverter.handleMacro: 39, 0 --> not a macro
> 14:59:39.397 -> KeyReport.handleModifier: 0, 0 --> not a modifier
> 14:59:39.397 -> KeyReport.removeKey: 39 --> removing from slot 0
> 14:59:39.397 -> KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]
> 14:59:39.397 -> 
> 14:59:39.397 -> suniversal: raw key 7f
> 14:59:39.397 -> suniversal: all released
> 14:59:39.397 -> KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]
> 14:59:39.397 -> 
> 14:59:39.397 -> suniversal: USB LED state changed: 3 --> 1
> 14:59:39.397 -> suniversal: SUN LED state changed: 9 --> 1
> 14:59:40.488 -> suniversal: raw key 62
> 14:59:40.488 -> suniversal: 62 down
> 14:59:40.488 -> KeyboardConverter.handleKey: 62 --> 53
> 14:59:40.488 -> KeyboardConverter.handleMacro: 53, 1 --> not a macro
> 14:59:40.488 -> KeyReport.handleModifier: 0, 1 --> not a modifier
> 14:59:40.488 -> KeyReport.addKey: 53 --> adding at slot 0
> 14:59:40.488 -> KeyReport.send: modifiers=0, keys=[ 53 0 0 0 0 0 ]
> 14:59:40.488 -> 
> 14:59:40.620 -> suniversal: raw key e2
> 14:59:40.620 -> suniversal: 62 up
> 14:59:40.620 -> KeyboardConverter.handleKey: 62 --> 53
> 14:59:40.620 -> KeyboardConverter.handleMacro: 53, 0 --> not a macro
> 14:59:40.620 -> KeyReport.handleModifier: 0, 0 --> not a modifier
> 14:59:40.620 -> KeyReport.removeKey: 53 --> removing from slot 0
> 14:59:40.620 -> KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]
> 14:59:40.620 -> 
> 14:59:40.620 -> suniversal: raw key 7f
> 14:59:40.620 -> suniversal: all released
> 14:59:40.620 -> KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]
> 14:59:40.620 -> 
> 14:59:40.620 -> suniversal: USB LED state changed: 1 --> 0
> 14:59:40.620 -> suniversal: SUN LED state changed: 1 --> 0
> 14:59:41.182 -> suniversal: raw key 62
> 14:59:41.182 -> suniversal: 62 down
> 14:59:41.182 -> KeyboardConverter.handleKey: 62 --> 53
> 14:59:41.182 -> KeyboardConverter.handleMacro: 53, 1 --> not a macro
> 14:59:41.182 -> KeyReport.handleModifier: 0, 1 --> not a modifier
> 14:59:41.182 -> KeyReport.addKey: 53 --> adding at slot 0
> 14:59:41.182 -> KeyReport.send: modifiers=0, keys=[ 53 0 0 0 0 0 ]
> 14:59:41.182 -> 
> 14:59:41.182 -> suniversal: USB LED state changed: 0 --> 1
> 14:59:41.182 -> suniversal: SUN LED state changed: 0 --> 1
> 14:59:41.348 -> suniversal: raw key e2
> 14:59:41.348 -> suniversal: 62 up
> 14:59:41.348 -> KeyboardConverter.handleKey: 62 --> 53
> 14:59:41.348 -> KeyboardConverter.handleMacro: 53, 0 --> not a macro
> 14:59:41.381 -> KeyReport.handleModifier: 0, 0 --> not a modifier
> 14:59:41.381 -> KeyReport.removeKey: 53 --> removing from slot 0
> 14:59:41.381 -> KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]
> 14:59:41.381 -> 
> 14:59:41.381 -> suniversal: raw key 7f
> 14:59:41.381 -> suniversal: all released
> 14:59:41.381 -> KeyReport.send: modifiers=0, keys=[ 0 0 0 0 0 0 ]
> 14:59:41.381 -> 
>

prosper00, 2021-02-01:
Here's a quick capture of what's happening on the TX (keyb Rx) line when I press, then release caps:

[image]

[image]

xelalexv, 2021-02-01:
The logic looks fine, suniversal correctly sees the CAPS LOCK activity, be it on the Sun keyboard itself or on your second keyboard, and it sends the according commands to switch LEDs on and off. It seems though that the commands sent from Arduino to the Sun keyboard have no effect. That's also why the log says at the very beginning that the layout cannot be determined.

What's the voltage level on pin 9 of your Arduino when there's no activity? Also, since you got a scope, could you capture some activity? And, what's the precise model number of your Sun keyboard?

xelalexv, 2021-02-01:
OK, you bet me with to the scope capture ;-)

prosper00, 2021-02-01:
keyboard is a 5c. Something is really odd though, the RX levels idle at 2V and transmit at about 2.5V, whereas TX idles at zero and transmits at 4.75V. When I disconnect the keyboard from D9, then the levels are correct. So, must be an electrical issue in the keyboard. Thanks for the help, I'm gonna see what else I can find....

Vintage hardware, eh?

closed 2021-02-01 by prosper00

xelalexv, 2021-02-01:
The bits sent to the Sun keyboard are correct: The first byte `0x0E` is the LED command, and bit 3 in the second byte controls the CAPS LOCK LED. I'm afraid there could be a hardware problem in the Sun keyboard.

prosper00, 2021-02-02:
Well, I found a burned up diode on the PCB. I don't know what it was, but replaced it with a 1n4148 and everything is working great now. It looks like a protection diode on the keyboard RX side. The PCB is discolored underneath it, and the mode of failure was such that it has a 2.0V reverse voltage and a 0.7V forward voltage..... very weird, but perhaps explains the 2V idle...

I wish I knew what it was that caused it to burn up in the first place, but, I can only hope that everything is fine now. My plan was originally to identify which pin on the keyboard processor was RX, and wire directly to that, and I suppose that's still an option if I have problems in the future. Tracing it out was how I noticed the bad diode.

xelalexv, 2021-02-02:
That's great! Happy to hear the keyboard is not dead.